easyeda/easyeda-documents #16: PDF

Spencerx, 2015-06-19:
Hi please can we share an official PDF of the simulation ebook.
Cheers

dillonHe, 2015-06-19:
Will think about this.

Thanks.

signality, 2015-06-19:
As previously discussed, I need to revise it anyway before we do anything
with it.

I might be able to generate a pdf easily from my original html version ....

Spencerx, 2015-06-19:
With an original source html file I can consider it a non issue.

dillonHe, 2015-06-19:
Check your email.

Spencerx, 2015-06-20:
Got it there are options to convert the documentation in one batch for the html archive.

I recommend 

htmldoc
Princexml http://www.princexml.com/download/

I'll get to that as soon as I can and I consider the issue closed.

closed 2015-06-20 by Spencerx

signality, 2015-06-21:
You can convert html to pdf using the Free Open Source Software, LibreOffice
<http://www.libreoffice.org/>.

Open with in LibreOffice Writer

File > Export to PDF