easyeda/easyeda-documents #29: Documentation typo cleanup

mMerlin, 2016-08-26:
I was looking at cleaning up some of the typos in the documentation, especially the Simulation eBook.  Simple spelling and grammer changes should be easy and safe, but is there a practical way for me to generate a clone of https://easyeda.com/Doc/Simulation-eBook from my repo?  This is using markdown files, but it looks like something more than github.io is needed to recreate the site content.

mMerlin, 2016-12-05:
Nudge.

Is this repo still active at all?

dillonHe, 2016-12-05:
Yes, but will be changed a lot soon for the new version

mMerlin, 2016-12-05:
How much is 'a lot'?  And would typo cleanup be useful at this point?

The original question still stands.  What is the process needed to generate the web site eBook from the repo?  To be able to actually view the results of any changes.  I can setup github.io on my fork, but is that enough?

dillonHe, 2016-12-06:
Too big, because the UI is not the same, almost the all of the images will
be charged.